FBurkelwsd/Eastlake-Shop #8: TA projects and notebooks

FBurkelwsd, 2024-05-01:
For us to document and capture progress, leave feedback and record future work as we optimize the Eastlake Shop. And help students with resources.

FBurkelwsd, 2024-05-01:
[image]

Maksym made this one

FBurkelwsd, 2024-05-01:
Next Steps: 
Getting Consumable materials ready for end of Year projects.
Areas to evaluate/make a plan/ensure enough information is labeled/ and make the area safer.

s-bitshaki, 2024-05-06:
Currently sorting through the storage room and getting to know the contents as well as the space limitations. Ideally we can sort it in a more efficient manner that allows easy access for all items and extra space for us to move the scrap metal bin.[image]

[image]

[image]

FBurkelwsd, 2024-05-07:
this table can go inside the back garage doors in D125 and using casters from shelving unit H in D124. Bolts/ nuts and washers, probably 1/4-20 are under the square table in D125.
[image]

FBurkelwsd, 2024-05-07:
<html xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:dt="uuid:C2F41010-65B3-11d1-A29F-00AA00C14882"
xmlns="http://www.w3.org/TR/REC-html40">

Need Help organizing Materials for the upcoming Automata box Project. Below are the criteria for success and the constraints. SHelving and scrap wood. Also to set aside space under tables and on shelving in D124 especially for projects.
<head>

<meta name=ProgId content=OneNote.File>
<meta name=Generator content="Microsoft OneNote 15">
</head>

<body lang=en-US style='font-family:Calibri;font-size:11.0pt'>

<div style='direction:ltr'>

Constraints:       Area needs to be        cleaned after each work day    Individual pieces need to        be stored in proper locations to help minimize clutter, and safety for        those passing by    Can not consume more wood        than a 30 cm square, unless taken from scrap materials.    Metal scrap and up to 3 vex        pieces can be used though need to be recorded.    Adhesives and in shop        hardware needs to be treated with respect and not wasted.    Materials can come from        home, as long as we have a place to safely store them. Please give        teacher notice.    Machinery will be loud,        remember we have ear protection on the wall.    Safety Glasses are by the        main door, should be on at all times when using any tool.    Guides for all machines        will be used as much as possible, feel free to ask a peer or the        teacher, though do answer tons of questions each day, making it        difficult to help everyone individually, though we strive to excel.    Feedback must be asked for        from peers, others in the students day, and teachers more than 3 times        throughout the project. | Put these in your   own words or images or memes here. Please label with arrow or number.
-- | --
Criteria for   Success:       Must have at        least 1 mechanical mechanism, respect and hard work will be granted to        anyone taking on a difficult project.    Student and teachers must        act professionally and safely while giving feedback, seeking help, using        tools, transporting, cleaning, and existing in our large shop area.    Custom features should be        added to any and all surfaces, allowing each student to showcase their        individual culture in a meaningful way.    We are striving for a final        mechanically moving final physical multimaterial project, which each        student can showcase in person and in a quick video explanation of        significant features.    CAD (fusion 360, onshape,        blender) images, files and videos of mechanical portions need to be        included in the design process. Respect and Hard work will be given to        anyone who has a CAD and Physical object that are identically uniquely        cultured mechanically and structurally. |  
ISTE AI    "It's great to hear that your students are working on a project that    involves the full design process, creating a functional computer model, and    building a final product like an automata box or mechanical toy. Several    ISTE standards align with this kind of project. Let's take a look at a few    that would be particularly relevant:         1.    **Computational Thinker (ISTE Standard 5)**: Students develop and employ    strategies for understanding and solving problems in ways that leverage the    power of technological methods to develop and test solutions. This standard    emphasizes leveraging technological methods to develop and test solutions,    which is directly applicable to creating a functional computer model and    building the final product.         2. **Critical    Thinking, Problem Solving, and Decision Making (ISTE Standard 4)**: This    standard involves identifying and defining authentic problems and    significant questions for investigation, collecting and analyzing data to    identify solutions and/or make informed decisions, and using multiple    processes and diverse perspectives to explore alternative solutions. These    skills are essential in the full design process and building a complex    product like an automata box or mechanical toy.         3. **Creative    Communicator (ISTE Standard 6)**: Students communicate clearly and express    themselves creatively for a variety of purposes using the platforms, tools,    styles, formats, and digital media appropriate to their goals. This standard    is relevant when students are presenting their design process, computer    model, and the final product.         By focusing on    these standards, students can not only engage in the technical aspects of    the project but also develop essential skills in problem-solving, critical    thinking, and effective communication, which are valuable in any    STEM-related project."         Chatgpt AI "    The ISTE (International Society for Technology in    Education) standards that align with this automata design project could    include:         ISTE Standard 4: Innovative Designer           Students          use a variety of technologies within a design process to identify and          solve problems by creating new, useful or imaginative solutions.      Students          develop, test, and refine prototypes as part of a cyclical design          process.          ISTE         Standard 5: Computational Thinker           Students          develop and employ strategies for understanding and solving problems          in ways that leverage the power of technologi [... cut ...] s (e.g., feedback from peers and teachers). This aligns with the standard's emphasis on understanding the ethical, cultural, and environmental implications of technology use.
By addressing these standards, your project not only provides students with hands-on experience in design and engineering but also fosters their understanding of the broader societal and environmental contexts of technology."

[Making an Automata : 11 Steps - Instructables](https://www.instructables.com/Making-an-Automata/)

FBurkelwsd, 2024-05-13:
We have updated the consumable material area well. For the south wall of D124. 

There is a Vex rolling horizontal cart that needs to have parts dissasembled and returned home. Vex Yellow drawers cart needs to be audited, large sign for above band saw wall can be made with 
[Uploading Shop norms 5.7.docx…]()

FBurkelwsd, 2024-05-13:
Do you want to be more involved in the slicing process? if so help iterate the printing que

FBurkelwsd, 2024-05-13:
What to do with TA catapult?

FBurkelwsd, 2024-05-20:
Organize and sort and labe and inventory in an excel spreadsheet. Miscellaneous electrical components on shelf 3 or d124 cart H. 
As seen in photos. 

Example. All breadboards into one labeled container and inventory. 

[image][image]

FBurkelwsd, 2024-05-20:
Need design and layout help for the recycling sorting and storage and electrics used for reverse engineering projects.plastics, cardboard, non aluminum metals, wood all types, electrical appliances, cords, broken electrical equipment. [image]![Uploading 332210482...]

FBurkelwsd, 2024-05-20:
Epilog laser. Screen is mounted. Needs a design for a mouse keyboard combination and NUke pc. 

Missing. Display cord from nuke to screen.

FBurkelwsd, 2024-05-21:
Inventory Update and Archive. 
[archive inventory eastlake 5.20.pdf](https://github.com/FBurkelwsd/Eastlake-Shop/files/15392099/archive.inventory.eastlake.5.20.pdf)
[Inventory May 24 update. .xlsx](https://github.com/FBurkelwsd/Eastlake-Shop/files/15392109/Inventory.May.24.update.xlsx)
Feel free to edit excel pages, with updated inventory information. We have a ton of material that is missing or has been recycled.

FBurkelwsd, 2024-05-21:
Measuring design project. [image]
[image][image][image]
Needs to make the many options apparent for choices and hoping to combine all into a wall or single area? Or split more evenly between two.

FBurkelwsd, 2024-06-24:
https://github.com/FBurkelwsd/Eastlake-Shop/assets/135162467/d5a501dd-e23f-4d39-968e-7caa29ca3f06

FBurkelwsd, 2024-08-02:
[image]
Trim this to fit under cnc

FBurkelwsd, 2024-08-28:
Goals!

D125 
Robotics club and class material scrap and management 

Rearrange for new layout. 

Areas to define: 
d125 shop door 
Wooden robotics structure and fire  alarm
Printer and cutter area
In front of sand blaster
South wall of d124 
Area in front of d124 door
Trailer area
Outdoor storage 

Common use spaces: 
d125 tables classroom and shop tables
Sink
Counter tops and cabinets
Consumables 
First aid
Wood materials
Aluminum
Machines
Trash and recycling 

Room map.