FBurkelwsd/Eastlake-Shop #13: Chop Saw

FBurkelwsd, 2024-05-02:
o	Avoid distraction.
o	Make sure the saw is securely attached to a table or base support before attempting any cut.
o	Hold or clamp the material securely before making any cut.
o	Keep hands at least 4 inches away from the rotating saw blade when making a cut.
o	Make sure the electrical cord and plug are in good operating order before using the saw.
o	After the saw blade has cut through the material, continue to hold the saw down, release the trigger switch, and let the saw blade stop rotating before allowing the blade to return to its starting position.
o	Never attempt to remove scrap material from the cut zone until the blade has stopped rotating and your hand is removed from the on/ off switch.
o	Make a trial cut on a piece of scrap material before attempting final cuts to make sure the saw is cutting properly and that the miter and bevel angles are set accurately.
o	Hold the material securely when making cuts.
o	When making multiple cuts of the same dimensions, clamp a stop on the saw table to assure accuracy of the cut and to reduce the time needed to make the cuts.
o	If the saw should kickback, continue to hold it firmly in the down position, release the off/on switch and wait for the blade to stop rotating.

Make sure ear protection is on and everyone near you is also on.

FBurkelwsd, 2024-05-02:
https://www.youtube.com/watch?v=h2UolfDERKM
https://www.youtube.com/watch?v=1UG_gUHvl90

FBurkelwsd, 2024-05-20:
[image]

vkirub, 2024-09-18:
Moved contents from here to https://ehsshop.packofparts.org/tools/chop-saw/ -

If you want to add more content, edit there

FBurkelwsd, 2024-10-03:
Making the physical machine usage signage better

vkirub, 2024-10-03:
Please use the git hub above for comments / edits.

FBurkelwsd, 2024-10-03:
cant attach a pdf of the file so here is the screen shot 
[image]